fpv-wtf/msp-osd #1: AU/Vista input voltage in OSD

SCMicroMike, 2022-04-04:
**Issue added to Margerine issues however was indicated that it would likely not be implemented at root tool level but maybe be taken up in an OSD mod. Bri3d feel free to kill this if this is not something your project would consider.**

As a user of the DJI FPV without a flight controller the display of the AU/Vista input voltage would be beneficial for use with the DJI FPV system.

The simple usecase is as follows:

Pilot installs AU/Vista unit in an airframe with direct flight battery connection according to hardware limitations.
Pilot powers on in AU/Vista and FPV goggles entering flight playback OSD.
According to value read and position values the flight battery is painted to the OSD.
The following comes from @funnel[](https://cdn.discordapp.com/attachments/950499167237718136/960576273665650738/unknown.png) on fpv.wtf Discord:

it's possible this is where the voltage value is in on the AU. not 100% sure, need to do more tests to confirm
cat /sys/devices/platform/soc/f0a00000.apb/f0a71000.omc/voltage4

[https://discordapp.com/channels/849931046883819550/950499167237718136/960576273841786910](https://github.com/fpv-wtf/margerine/issues/url)

multiplied by ~0.01584

yes it is. I went through 13-16.8V with a psu and that's where you can read it. The average factor on my old big AU is 0.015833
I don't have a Vista to test

bri3d, 2022-04-04:
This could be implemented as an additional feature on the air side in one of several ways:

* Read the input voltage and send a DJI DUML message. This could even be implemented using a shell script and the `dji_mb` commands in `/system/bin` provided we find the correct message to send. Benefits: Shows up natively in DJI Glasses OSD (so warnings etc. work), should be pretty easy. No tampering with the system, can probably be done without even writing any native code. Drawbacks: Maybe the message we send will conflict with HDVT or get overridden? And need to find the message to send.
* Send a fake MSP "reply" over the serial mux to the DJI HDVT service, populated with the voltage read from the input pins. This also should be pretty easy. Benefits: Same as previous method, plus only relies on things we know (MSP and the mux service) instead of researching things we don't (DUML message for voltage). Drawbacks: Requires the serial mux service to be installed and run, so a little "heavier weight" in terms of tampering with the system.
* Send an MSP OSD with only a Voltage widget, populated by reading the voltage. Benefits: 100% in code we control. Drawbacks: Requires you to kill `glasses` on the goggles side and lose all DJI functionality to get only a voltage display. Heaviest weight in terms of tampering with the system.

I think approach #2 is the best - basically, extend the serial mux with a "fake flight controller" function which can reply to various MSP queries with data populated off of the Vista/AU. The existing DJI OSD functionalities can be used.

I'll take a look at this later, I had no idea the air side stuff even had voltage sensing so I'll play around.

bri3d, 2022-08-22:
This has been done for awhile now.

SCMicroMike, 2022-08-22:
Nice!! I will look at the project and see if I can get a wing up and running to test it.

bri3d, 2022-08-22:
Oh, actually, I do need to make a small fix for use without FC altogether,
as right now the FC triggers the on screen drawing. I’ll update when this
is done.

SCMicroMike, 2022-08-22:
Copy that. Will be on the lookout. Thanks for addressing this. I have a couple questions but will ask on the discord to not clutter this.

driedger, 2022-09-27:
Following this issue, and request that it gets reopened to ensure it doesnt fall through the cracks. 
Would be great to modify the au_overlay_enabled option, or create a new option on the air unit side, to send temp and voltage without having a fc detected. As an AU option it could be enabled only as needed, 
In the code it appears this only fires when we process_data_packet, which only happens while receiving msp data. @bri3d I suspect you already have a plan on how to implement this so I will stop digging though the code!
Thanks

bri3d, 2022-10-08:
0.6.11

closed 2022-10-08 by bri3d