fpv-wtf/msp-osd #3: fonts colour and inav support ?

singlag, 2022-04-11:
can add different fonts colour and support for inav ?
such as "inav radar"

bri3d, 2022-04-15:
Closing this out as you added support for font color to mcm2img.

closed 2022-04-15 by bri3d

SilentZen, 2022-04-15:
> iNav works fine, use https://github.com/bri3d/mcm2img like: `python3 mcm2img.py mcmfile.mcm font.bin RGBA` to generate a new font file from an iNav MCM font.
I have generated new font but I cannot get OSD to work like on betaflight in inav, I just get Waiting for data in OSD, I have tried HDzero OSD on the uart connected to AU and also FrSkyOSD. What is the trick to getting this to work on iNav?

bri3d, 2022-04-15:
HDZero worked for me, do you have dji_hdvt_uav running on the AU? The HDZero displayport doesn't activate unless it sees inbound MSP traffic on the serial port. It's actually a better implementation than the normal Betaflight one and the update rate should be better.

FrSkyOSD is totally different protocol (pixel canvas) and won't work, it requires a much too high bitrate.

SilentZen, 2022-04-15:
Thanks that's exactly what I missed "setprop dji.hdvt_uav_service 1"