fpv-wtf/msp-osd #4: Autostart after reboot

DimitarKrastev, 2022-04-11:
It is a bit unclear if I need to hook the Air Unit/Goggles to the PC and start the process every time after the installation.

If I reboot the Air Unit/Goggles, do I need to manually start the process every time?
Is it possible to register it as a service on the devices and switch between the regular UI and the MSP DisplayPort by just using the Back button after reboot?

Thanks,
Keep up the great work!

j005u, 2022-04-11:
We're working on an update to margerine to allow for registering of custom services, it'll be out soon(-ish). Right now it requires manually patching the official startup scripts, which can be dangerous.

DimitarKrastev, 2022-04-11:
Great, are you looking for contributers? I am a senior software engineer, but I will need a quick onboarding with the android and DJI specifics. If you are looking for contributers, are you willing to have a call?

j005u, 2022-04-11:
@DimitarKrastev have a look at the various margerine [wiki pages](https://github.com/fpv-wtf/margerine/wiki) for a first introduction, including the ["todo list"](https://github.com/fpv-wtf/margerine/wiki/Possible-Features) to get an idea what the opportunities are.

I'm close to being done with package management (opkg/entware) and service management (dinit) and should have a release out soon-ish. Happy to have a chat after I have that out the door.

bri3d, 2022-07-08:
Fixed by WTFOS release

closed 2022-07-08 by bri3d