fpv-wtf/msp-osd #10: ready to use binaries?

Blueforcer, 2022-04-24:
I have massive problems setting up the build enviroment and cant get it to work :(
Is it possible to upload a release or maybe someone can build it with a backup of my system/lib ?

i would really appreciate

singlag, 2022-04-26:
> I have massive problems setting up the build enviroment and cant get it to work :( Is it possible to upload a release or maybe someone can build it with a backup of my system/lib ?
> 
> i would really appreciate

maybe you can try this one, build for V1 goggle (.600) and airunit (.600)
https://github.com/bri3d/msp-osd/issues/7

Blueforcer, 2022-04-26:
> > I have massive problems setting up the build enviroment and cant get it to work :( Is it possible to upload a release or maybe someone can build it with a backup of my system/lib ?
> > i would really appreciate
> 
> maybe you can try this one, build for V1 goggle (.600) and airunit (.600), contain windows batch file to patch auto startup #7

meh, forgot to mention that i have a v2 :-)

closed 2022-04-28 by Blueforcer