fpv-wtf/msp-osd #12: Automatic BF setup

j005u, 2022-05-14:
It would be awesome if the air side mux could set up MSP OSD for you automatically in BF. 

I guess some probing to find the correct MSP uart might be involved.

bri3d, 2022-05-14:
I had never thought of this hack idea and I kind of like it. Very BF
specific though but maybe I can get platform / font auto detection out of
this too if I do it right

j005u, 2023-05-22:
Betaflight now has a way to differentiate between a generic MSP device and one set up for a VTX: https://betaflight.com/docs/development/Serial#2-serial-port-function

Meaning for BF 4.4 users we could figure out that they have a single MSP VTX type device configured and re-configure it to DP automatically.

This could help avoid confusion with users that don't RTFM and never turn off DJI's custom OSD and therefor never finish correctly configuring DP on the FC side.