fpv-wtf/msp-osd #13: Update rate nosedives off a cliff

j005u, 2022-05-15:
The update rate seems to be pretty good for a 2-3 seconds after which it slows down significantly.

Why?

bri3d, 2022-05-15:
This is because the `dji_hdvt_uav` process is also spamming the MSP processing logic with a gazillion MSP messages to send back over DUML. 

I tried increasing the baud rate (the `-f` option in the OSD mux) thinking it was line saturation on the UART, but it's not, it's scheduling saturation on the MSP+serial tasks. 

This can be proven/confirmed by stopping dji_hdvt_uav once the OSD is running - you will observe that the update rate recovers immediately. Of course, then you lose channel change, DJI telemetry, and arm/disarm monitoring at minimum, which is definitely undesirable. 

What I need to do is write a "Fake FC" shim which sits in front of the map mux and responds to dji_hdvt_uav with fake/proxied data so that we're not letting DJI blast the MSP port.

bri3d, 2022-07-08:
Fixed.

closed 2022-07-08 by bri3d