fpv-wtf/msp-osd #19: Random relocation of OSD elements

fabriciopolonio, 2022-06-07:
When using the msp-osd, sometimes some of the OSD elements blink on the screen on top of other existing elements, for example:

Let's say on my OSD I have:

* Remaining battery (top)
* Craft name (left)
* Link Quality (left)

Sometimes the "Remaning battery" element for a short time is displayed on top of the "Link Quality" or the "Craft Name", then it goes back to its set position.

bri3d, 2022-06-08:
This is quite strange, I will look into seeing if I can reproduce this issue.

This is odd because MSP commands are processed command-wise and are checksummed. It's odd that drawing could happen in the wrong place.

I need to add a debugging functionality (which I have been working on as part of overlaying OSD in post-processing anyway) which will dump the MSP frames received by the goggles to figure out what could be going on here.

fabriciopolonio, 2022-06-10:
@bri3d if you need I can share my Betaflight OSD setup, in fact it happens in every flight I perform.

DimitarKrastev, 2022-06-10:
@fabriciopolonio Check your wiring. I started seeing this in the standard DJI goggles firmware with their stock limitted OSD. I started seeing the telemetry disappearing randomly for fraction of the second and random elements popping up in different parts of the screen. Even elements that I have not enabled like the bar battery indicator. This started happening on my last flight and I have crashed my drone a few times so I suspect something with the wiring.

bri3d, 2022-07-08:
The latest version of msp-osd may help with this issue. Previously there was an issue where the byte stream could have ended up shifted by one byte, which may have led to the first or last message in a "frame" getting dropped randomly.

bri3d, 2022-07-16:
I think this should be fixed

closed 2022-07-16 by bri3d