fpv-wtf/msp-osd #23: HD OSD on non Inav FC?

DimitarKrastev, 2022-07-15:
I noticed a recent commit with INAV HD OSD. How does that work in general and how can this be enabled in BetaFlight and ArduPilot?

Thanks and great work!

DimitarKrastev, 2022-07-15:
Great, thanks. I have a proposal for similar implementation  open in another issue:
https://github.com/bri3d/msp-osd/issues/16

Basically the idea is to use the coordinates as indexes multiplied by a given factor. That should give more effective resolution even if the sending side doesn't support that. I think DJI is using something similar as approach as their OSD is much higher resolution and "zoomed out". The only downside to it is the addressability of the position of the elements. If we draw twice as many character boxes in the OSD, but the sending side only know about half of them it would mean that moving an OSD element 1 box to the right will result to 2 boxes of movement in the goggles, but I think thats a small price to pay.

Can you take a look and let me know what you think?
Thanks!

TomAlperin, 2022-09-25:
I was thinking something like this would be a good idea. It would be like if Fake HD had space between all the cells so they filled the screen but the cells would only set the starting point. The additional letters would be the correct spacing from the previous. Fake HD makes it difficult to position some long elements without them being split.

benlumley, 2022-09-26:
This approach isn't really that practical for various reasons (eg: inav likes to do partial updates; BF sends each character individually, some elements span multiple lines - nothing would line up, some elements grow/shrink depending on their data). 

I am intending to add some configurability to the grid - so you can have space for longer elements; which should solve your problem here.

closed 2022-10-24 by benlumley