fpv-wtf/msp-osd #26: Msp-osd in Inav

Ben-Stevenss, 2022-08-06:
In the ports tab you have to select hdzero, does the msp also need to be switched on?

somehow i can’t seem to switch msp and hdzero on together. 

I have the osd working/running but the elements are wrong/messed up. 
and yes i tried going in the menu with stick command and back.

bri3d, 2022-08-13:
No, MSP should not also be switched on, just HDZero VTx.

What fonts do you have on your SD card, if any?

Ben-Stevenss, 2022-08-13:
Impact mini renamed to font. 

But the one from the betaflight folder. 
Did a couple of betaflight quads, and a few days later tried my inav wing. Can try the inav font?

bri3d, 2022-08-13:
Yes, this will be why, then. Need to use font corresponding to FC in use. I will add a feature to ID the FC and use different fonts at some point soon here.

Ben-Stevenss, 2022-08-13:
So at the moment i can't use betaflight and inav together?

bri3d, 2022-08-26:
One trick for this is to use iNav in HD mode, and use an iNav font as the HD font (font_hd and font_hd_2) on the SD card. Then leave your "normal" font as a Betaflight one.

bri3d, 2022-08-26:
Down the road, I plan to add FC detection to determine which fonts to use.

closed 2022-09-21 by bri3d