fpv-wtf/msp-osd #29: OSD low update rate or frozen

Jordi1990, 2022-08-12:
I'm seeing an issue where the OSD is not updating anymore or very slow. This occurs about every flight.

What I observed:
- OSD elements frozen(few seconds or as long as a few minutes), eventually recovers
- OSD elements updating really fast after being frozen for a short while
- Might be triggered by a bad signal, need to do more tests to confirm that.

System info:
iNav 5
MSP-OSD 0.5.1

Let me know if any other information is needed, I would also be able to run a debug build if required.

bri3d, 2022-08-13:
Thanks for letting me know.

Can you try without boost-bitrate and see if this happens?

I'll run some bench tests myself and then some flight tests and see if I can reproduce.

Jordi1990, 2022-08-13:
I just flew a single pack with boost-bitrate removed. Still had an occasional freeze. I could reproduce it consistently by letting the signal get so weak that the feed starts stuttering. While the signal is that bad the OSD is completely frozen and when I punched out it immediately restored.

However this flight I didn't see the behavior where it stayed frozen for a longer time, last time this occured 

What might be interesting is that also the bitrate and latency are frozen at the time.

bri3d, 2022-08-13:
OK, this is logical. Bitrate and latency are sent over UDP also. The issue here is just that the link is degraded to the point that UDP packets aren't getting from the airside to the goggles. Not much I can do about this without totally changing the transmission method.

DimitarKrastev, 2022-10-27:
As far as I know the boost bitrate is probably something you don't want to use. While boosting the bitrate seems like an obvious win, the way it is achieved is by disabling certain link-stability features. 

The result would be:
* Better link when you have a strong signal
* Much worse link when the signal gets weak

I would say what you experience is by design.

tvtechdir, 2023-04-12:
Just adding a bump to this thread with some varying circumstances. I'm running Betaflight 4.3.0? on a 7" long range with an older Diatone Mamba F7APP FC. Last week I encountered complete OSD freezes on the second and third of three flights which were all launched from above 8,000' ASL. The first time I was only a mile or so out. The second I lost video behind mountain peak two miles out. Once line of sight was restored video came back but in both cases the OSD never revived until the quad was power cycled. Will update this VTX when I get home and see if I can duplicate.

Caddx Vista, not sure the wtfos version (was installed sometime in late December/January)
Goggles V2