fpv-wtf/msp-osd #36: missing a FAQ section - makes me ask here.

AndKe, 2022-08-27:
Last I heard, this great hack required a manual start on every boot.
-Is a persistent mode possible now? / soon?
-what's the last/best safe version one can upgrade DJI goggles/Caddx Vista to ? - I don't dare go beyond 1.00.0606
-are there any adverse effects of this hack?

Thank you - keep up the great job.

bri3d, 2022-08-29:
- Just install using testing.fpv.wtf . Everything automated. msp-osd will be part of your DJI goggles experience.
- .0606
- Right now, there's no toggle, so you need to reboot your goggles if you switch to a quad/plane without MSP-OSD, or the OSD stays up. Also, it doesn't auto-dismiss during analog in or playback yet. These are probably the only major minuses I can think of. You can always disable using testing.fpv.wtf to "go back" if this is a big issue for you.

closed 2022-08-29 by bri3d

AndKe, 2022-08-29:
Thank you.
I'll definitely test it now.

One more thing I've always missed: When flying on long-range system, the FPV link is likely to fail long before the RC control.
Have you thought of a feature where the Caddx Vista could trigger "GPS RESCUE" of Betaflight if/when FPV video fail? (and possibly: if distance is greater than a preset number)