fpv-wtf/msp-osd #37: Font dual use for inav and bf

robschii, 2022-08-28:
When I choose a font it is not working for bf and inav. Only one is working.
Please do 2 seperate folders on the goggle for inav and  bf.

FPVZaphod, 2022-09-24:
Yes, folders for each Software would be cool, to manually choose in a menu?
Maybe it's even possible to remember the setting, based on a serial number of the air unit?

benlumley, 2022-09-26:
Right know msp-osd doesn't know if it is working with inav/bf/whatever....  so whilst the code supports different variants of fonts; until we get a way to do switching we can't go further. 

Short term - either :

 - Use SD for BF + HD for INAV, with an SD BF font and an HD inav one.
 - Use FakeHD for BF, HD for INAV - when the code detects RealHD it force disables FakeHD. Since FakeHD uses the bf variant (font_bf_hd*.bin) and INAV uses the generic variant (font_hd_*.bin), you can also switch this way.  (This is new to the latest version)

closed 2022-10-16 by bri3d