fpv-wtf/msp-osd #42: Problem with operation on DJI V2 Goggles

Hwurzburg, 2022-09-03:
Rooted the goggles, using the WTF config, installed the WTF-OS, added the msp-osd.....getting the waiting for OSD message in goggles...

was successfully using ArduPilot to display custom OSD data using the SERIALx_PROTOCOL 33 before....switched to 42, OSD_TYPE=5, and options for using Betaflight default font....

nothing....no display....turn off msp-osd, revert to SERIALx_PROTOCOL = 33 and OSD_TYPE=3 (MSP) and osd displays again,  with the normal custom osd limitations....

any suggestions as to what to try? or how to debug...

bri3d, 2022-09-04:
Did you install msp-osd on your air unit / vista as well?

Hwurzburg, 2022-09-12:
duh...working now...

closed 2022-09-12 by Hwurzburg