fpv-wtf/msp-osd #43: Vista goes into Low Power Mode as soon as MSP-OSD is enabled.

TestPilotIan, 2022-09-03:
Got a funny problem with my setup, so whenever I switch on the MSP-OSD in the Configurator on my Vista, it instantly (or within a few seconds) goes into low power mode. even thought Auto Temperature Control is set to OFF in the Googles menu.  This also happens upon power up, within a few seconds of everything syncing up, the vista goes into LPM regardless of setting.  This is a brand new Vista on 0606 and I had the same problem on the older, beaten to within an inch of it's life only transmitting at 5Mb/s unit it replaced.

Arming the quad allows it to go to full power and recording starts fine, but at our club we do a lot of crashing so don't want the Vista unit going into low power mode when disarming (this does happen) because then often we cannot turtle or get out of the trees, some of us also like to start recording before arming to sync up separate audio recordings.

Strangely, if I go into my googles menu and change Auto Temperature Control to ON, then turn it back to OFF again, the vista unit then respects this setting and will happily sit there until you get the Overheating please land message you usually get if you let them overheat.

I asked on the FPV.WTF discord and they said to ask here, is this a feature and if it is, can it be disabled, or are my crappy old v1 googles (that are stuck on 0600 firmware because every time I try to upgrade DJI Assistant says no firmware is available) causing the issue, if so do you recommend trying to upgrade them to 0606 or is this a known issue and I just need to toggle the Auto Temperature Control setting everytime I fly?

Also, any chance of getting the vista's mw output onto that list that has the temperature and voltage in on the osd?  We're a bunch of idiots at our club and will often take off in 25mw mode and the first anybody knows that they're not in 700mw power is when you're power looping a big tree and loose video on the other side of it LOL.

Any help or pointers much appreciated, thanks.

bri3d, 2022-09-04:
Definitely not a feature. Quite strange really that this correlates with enable/disable on the Vista side. It must relate to restarting the `dji_hdvt_uav` process on the Vista.

I haven't done any research into how low power mode is sent/received/configured, so I'd have to look into this a lot more to tell you what's going on, but it's surely not intentional.

re: updating your goggles, you need to disable ADB, the DJI configurator doesn't like V1 goggles with the ADB flag toggled on. There's a script to do this, `wtfos-remove-adb` . To update your V1s, uninstall WTFOS, go into the CLI tab, and run `wtfos-remove-adb` . Then update in the DJI Assistant app, and then re-root and reinstall WTFOS.

TestPilotIan, 2022-09-04:
Thank you, I will try to get a video for you so you can see what is happening.

Thanks.
Ian

grimr33f3r, 2022-09-18:
Also seeing this happening, on iNav though doubt that has anything to do with it.

Everything (Goggles v2 & quite an old Caddx Vista Nebula Pro) freshly updated to latest firmware, rooted & WTFOS installed immediately afterwards. Was all fine prior and uninstalling msp-osd stops it from happening. Everything else working completely fine.

It seems to be alright for a few seconds after starting up, but once the UI refreshes briefly with the voltage for the Vista, that's when it goes into low power and needs a toggle of the Auto Temp to kick it back out (Haven't tried arming yet as noticed it when bench testing after rooting).

Toggling the Auto Temp control serves as a workaround in the meantime.

Cheers!

evensis, 2022-09-18:
Changing the output seemed to help (a bit), in that 1 / 1.2w would guarantee it go into low power mode, and anything lower from 700mw down, it would be fine for a little while then it would drop off (here just in case it's relevant to diagnosing the issue). Setting 25mw seemed consistent and didn't go back into lower power mode, for the 2-3 minutes I was testing that power output trying to diagnose this. I was using bit rate to determine LQ and the model was in the open air at the local flying strip.

On my goggles at least, the low power mode message in the low right-hand corner, was NOT present (it was initially and then disappeared) which would make sense, as it was armed. Consequently, I fear this issue has a propensity to lull users into a false sense of security in regard to the quality of their video link. At least, I was in the air before I noticed this issue which was the worst possible scenario, and had to dust off my line of sight skills in a hurry.

TestPilotIan, 2022-09-19:
I’m guessing that a lot of people run with auto temp control on so this behaviour won’t be anything odd to them, it’s only if you’ve been running with it off for a long time and got used to it never going in to LPM that you notice when it does.

Cheers
Ian

bri3d, 2022-09-20:
Can you check whether this Low Power issue still happens after a full WTFOS upgrade? Joonas did some work to prevent the module injector (modloader) from causing process restarts across the reboot. I think this _should_ fix the issue, unless it's with the process restart on the airside.

I can't reliably reproduce this anymore but I also couldn't get it to happen reliably to start with, so I'm a bad test candidate.

grimr33f3r, 2022-09-20:
Just done full a update on Vista & goggles, it's still using Auto Temp Control when set to off and going into Low Power Mode.

For clarification, it correctly unlocks when armed (Both iNav & BF), it's just ignoring the setting.

bri3d, 2022-09-20:
Thanks for checking. Does the Auto Temp Control menu item say Off but it's still in low power mode, or does the menu item also toggle back to On when you didn't want it to?

grimr33f3r, 2022-09-20:
It says Off, even with Low Power mode active. That's the part that threw me at first as I assumed that the UI was correct.

bri3d, 2022-09-20:
Aha! Thanks. That's a good pointer and helps me follow the code around. I will continue debugging.

TestPilotIan, 2022-09-20:
Yup can confirm above with mine too.

bri3d, 2022-09-21:
Thanks for the bug reports and detailed reproduction steps. Really helpful when debugging!

I've found the root cause of this issue: as part of the startup of `msp_displayport_mux`, we have to restart `dji_hdvt_uav` to re-direct the serial port to the virtual one created by the displayport mux. 

However, `dji_shuttle` is responsible for loading the user preferences on the Vista/AU, and it doesn't re-send this specific preference to the `hdvt_uav` process after that process restarts.

I believe this can be fixed by just restarting `shuttle` in addition to `hdvt_uav` .

bri3d, 2022-09-21:
Hoping to have a fix out soon.

TestPilotIan, 2022-09-21:
Awesome thanks.

commit f3239158 (2022-09-21): restart shuttle service, hope to fix #43
PR #53 (2022-09-21): Add shuttle restart
Fix #43
commit fdd19723 (2022-09-21): Add shuttle restart, fixes low power mode issues (#53)

* restart shuttle service, hope to fix #43

* 0.6.5
closed 2022-09-21 by bri3d