fpv-wtf/msp-osd #47: Save framebuffer to file.

cvetaevvitaliy, 2022-09-08:
Hi! Very cool project, thanks to everyone who contributed!

I have an idea, what if we take the data that comes into the framebuffer and save it ?
This will allow us to have video with OSD data.

example save framebufer to file 
`ffmpeg -f fbdev -framerate 25 -i /dev/fb0 output.avi`

I see package `ffmpeg` utility  in Package Manager

bri3d, 2022-09-08:
A few notes:

* The framebuffers used in DJI are never exposed to the Linux `fbdev` framebuffer infrastructure - they're all allocated internally and managed through the DJI `libduml_hal` driver. 
* Video with OSD data composited onto it never exists in a framebuffer that is exposed in any known way. The video stream from the VTx (or playback) is decoded in the RTOS running on the CP (coprocessor) CPU and is sent directly to the screen from there. Overlay layers like the DJI menus and OSD are produced on the AP (Application Processor) CPU and sent to the CP (Coprocessor) CPU using shared memory and ICC (Inter Core Communication).
* Based on disassembly of the Coprocessor RTOS, it seems that the layers are actually blended in hardware at the LCDC (LCD Controller). So, as far as we know a composited screen buffer - that is, a set of pixel data that includes both the video feed and the OSD, doesn't exist in RAM that's accessible to the CPU at all. One experiment here could be to dump all of RAM to see if the LCD Controller also has a shared RAM area that perhaps it is using as a render target, although it's equally likely that it has a final framebuffer which is internal to the LCD Controller. 

My intended solution to this is to dump out the MSP OSD data into a file while video is being recorded (the main issue here is time sync, which is annoying but certainly a solvable problem). 

Once the MSP OSD data is in a file, we can write a PC application where we have plenty of CPU and encoding power which can draw the OSD again, composite the video, and render it out into a new file.

cvetaevvitaliy, 2022-09-09:
> 

Hi! Yes you’re right in the frame buffer does not store the image that is transmitted from the AirUnit.
I added libpng to the project and saved the current frame to a file, there I see only OSD data without an image from the AirUnit.

And I thought, maybe attach a screenshot to the push Back button 

[image]

cvetaevvitaliy, 2022-09-09:
> 
but about saving MSP data to a file, it's possible.  The trigger to start recording can be an ARM command in an MSP frame, goggle also start recording after ARM on, synchronization can be done manually (maximum 0.5-1s desynchronization)

cvetaevvitaliy, 2022-09-09:
Test saved MSP data from from DJI FPV
[data.msp.zip](https://github.com/fpv-wtf/msp-osd/files/9538985/data.msp.zip)
A little later I will amke commit and PR to your repo

j005u, 2022-09-09:
@cvetaevvitaliy mine is /mnt/media_rw/08F6-9CDC

Use the /storage/sdcard0 symlink instead. You can also check for the presence of the link to make sure the sdcard is mounted.

Protip: you can use `setprop persist.dji.storage.exportable 0` do disable unmounting the SD card from the Goggles while connected to PC for adb.

FPVZaphod, 2022-09-24:
It would be cool already to save MSP Data to a file, and then having a tool to render an OSD out of it, and overlay it to the video afterwards. Luxurity would be a 2nd video file, only with OSD, to just mix them afterwards :-)

bri3d, 2022-10-16:
Knifa added MSP data saving a few versions ago

closed 2022-10-16 by bri3d