fpv-wtf/msp-osd #65: msp-osd looses fonts on reboot when using fakehd or fakehd settings not saving....

TestPilotIan, 2022-09-24:
Got a strange issue, with msp-osd when using fakehd.

This problem happens with all fonts, so I put the fonts onto my googles and named them correctly, the fonts don't show and the regular font displays as normal.  If I then go to the cli and type package-config set msp-osd fakehd_enable true and press Enter, then package-config apply msp-osd and hit enter again, pigeon says it's saved the settings and reboots msp-osd - the fonts then show correctly, they look fab.

Then if I power cycle my googles, it's back to square one, the font used is the original font that shows when you don't have any fonts on the sd card, if you then goto the cli and type package-config set msp-osd fakehd_enable true and apply msp-osd the font starts working fine again, until the next power cycle.

Sorry the pic is out of focus, but this is cold boot font:
[image]

and this one is the nice font shown after sending the enabling fakehd command and saving.
[image]

My setup is latest build, betaflight 431.

Is this a bug?

benlumley, 2022-09-25:
can't see in the screenshots; is the cold boot one back to sd ? or is it fakehd with the wrong font ? 

does your SD card work reliably? have you tried a different one ?

TestPilotIan, 2022-09-25:
To be honest, I’m not really sure, the layout stays the same regardless.  When I tried fakehdthe other day (on the old build) it spread out like it’s supposed to do, this one doesn’t.

My friend took a photo today at the field I’ll post it tomorrow for you.  In meantime I’ll try another SD card, it’s never been a problem in the past but you never know.

TestPilotIan, 2022-09-25:
Here’s the pic, it looks like it goes back to no HD setup.
Also there is this strange black line to the right of both the voltages, guessing that’s the missing v?  The logos also were all wrong today jumping between things like  kph and other glyphs instead of the battery symbol.

Hope that helps

benlumley, 2022-09-26:
betaflight, yeah ? 

if it's going back to SD; then it seems like settings aren't saving.
if it stays at FakeHD and just misses the font; then it could be that the SD card doesn't mount initially.... 

Not super sure; learning here too...

benlumley, 2022-09-26:
After a fresh boot; can you run this command and copy/paste the output please?

```
package-config get msp-osd
```

TestPilotIan, 2022-09-27:
Hi the ouput is:

root@pigeon_wm150_gls:/ # package-config get msp-osd
show_waiting='true'
show_au_data='true'
fakehd_enable='true'
fakehd_hide_throttle_element='false'
fakehd_lock_center='true'
root@pigeon_wm150_gls:/ # 

I will check another SD card later today.

benlumley, 2022-09-27:
if you want it to spread out, you need to turn off `fakehd_lock_center`

TestPilotIan, 2022-09-27:
OK so I think it's me.  I was trying to use the iNav font which is much nicer, but msp-osd only loads that font to use if you send the  package-config set msp-osd fakehd_enable true [Enter] then package-config apply msp-osd [Enter] then the msp-osd proces restarts, and loads up the iNav fonts.

If I put any of the regular betaflight fonts in as font_bf_hd.bin / font_bf_hd_2.bin, they work fine.

So this can be closed, but maybe it can give a clue as to how msp-osd can be modified to use the hd fonts inav's system uses.

Cheers

TestPilotIan, 2022-09-27:
I checked another SD card btw, so it's not that.

benlumley, 2022-09-27:
OK - so that makes sense.

The SD card gets unmounted from the goggles and attached to your computer instead when you plug in USB (ie: it's invisible to the goggles)

`package-config apply msp-osd` cause the UI to restart, and re-read the fonts.... at which point because the SD card has gone; it can't find those and falls back to the ones built into msp-osd.

Check out the latest version; it has flight controller now and picks fonts based on this. It should be in the docs, but isn't yet: https://github.com/fpv-wtf/msp-osd/pull/68

closed 2022-09-27 by benlumley

TestPilotIan, 2022-09-28:
It’s actually the other way round, when msp-osd restarts it loads the hd fonts correctly, on cold boot it reverted to the inbuilt fonts.

My theory is the issue is the iNav HD font I was trying to be clever with and using won’t load until the process is restarted, it’s probably a bug that it actually loads up the font and it works when restarted more so than it being a bug by not working at a cold start.  As I said all the Betaflight based fonts work as intended and designed if I use them, it was only the iNav one I nicked off the msp-osd off the source page that I was having this issue with, shame as it’s a nice font, but I can live without it.

Cheers
Ian