fpv-wtf/msp-osd #70: fakehd_lock_center

gavris07, 2022-09-28:
package-config set msp-osd fakehd_lock_center true

 does not work

All the same, gaps in the OSD, but I don't need them

bri3d, 2022-09-28:
It's unclear what you are saying.

Can you paste the output of `package-config get msp-osd` ?

gavris07, 2022-09-28:
Here's the answer
root@pigeon_wm150_gls:/ # package-config get msp-osd
show_waiting='true'
show_au_data='false'
fakehd_enable='true'
fakehd_hide_throttle_element='false'
fakehd_lock_center='true'

gavris07, 2022-09-28:
after reloading the glasses gives this
root@pigeon_wm150_gls:/ # package-config get msp-osd
show_waiting='true'
show_au_data='false'
fakehd_enable='true'
fakehd_hide_throttle_element='false'
fakehd_lock_center='false'

gavris07, 2022-09-29:
made a reroute, the glasses began to save the settings, the topic can be closed))

closed 2022-09-29 by benlumley