fpv-wtf/msp-osd #82: fakehd_hide_menu_switch do not work

druckgott, 2022-10-23:
Hallo, I aktivate:

```
package-config set msp-osd fakehd_hide_menu_switch true
package-config set msp-osd fakehd_menu_switch 4
package-config apply msp-osd
```
In Betaflight I aktivate: 
[image]

But If I disarm, the overview from betaflight is not fit to the middle.
I have msp-osd current version installed. (V0.7.2)

benlumley, 2022-10-23:
Is the throttle element on the osd layout in betaflight?

druckgott, 2022-10-23:
How does this look like or how can I switch it on?

benlumley, 2022-10-24:
[image]

druckgott, 2022-10-24:
Ok I changed it like this:
package-config set msp-osd fakehd_hide_menu_switch false
package-config set msp-osd fakehd_menu_switch 137
package-config apply msp-osd

And used the GPS Globe because I did not have Throttle.
[image]

Now it´s working, maybe it makes sence to put this example to the readme.

closed 2022-10-24 by benlumley