fpv-wtf/msp-osd #83: Update docs + preset ahead of bf 4.4 release

benlumley, 2022-10-26:
https://github.com/betaflight/betaflight/pull/11913

No more `displayport_msp_serial` command; it's in the configurator as a peripheral now; (part of the function param of the serial config)

benlumley, 2022-11-08:
yeah... i don't know the command. 

the nightly configurator can do it on the ports tab you can choose "OSD (MSP DISPLAYPORT)" for the uart in question

The cli command might be:

```serial X 131072 115200 57600 0 115200```

Where X is the 0 indexed (0 = uart1, 1 = uart2 etc) uart port number. But i've not tested it. 

(in place of the older set displayport_msp_serial = <ConfiguratorUART - 1>)

So...

```
set osd_displayport_device = MSP
serial 0 131072 115200 57600 0 115200
save

```

medved91, 2022-11-08:
@benlumley thanks! Asked about cli command because failed to find `OSD (MSP DISPLAYPORT)` setting and it turns out that there isn't any in latest builds!

Look, they've merged `OSD (MSP DISPLAYPORT)` with `VTX (MSP)` into `VTX (MSP + Displayport)`

https://github.com/betaflight/betaflight-configurator/commit/4b3f1dbb6a2717625b5b7a48fd9e3d0716f56a31

benlumley, 2022-11-17:
Support for the new BF 4.4 features is being worked on; but hasn't arrived yet.

For now; select PAL/SD + use FakeHD is the best option for BF 4.4

MSP-OSD should support the newer negotiated size supported from BF 4.4 within a few weeks; it will likely negotiate to 60 * 22 as that fills the display nicely at current font sizes.

davidchriqui, 2022-12-28:
On BF 4.4, when enabling fake HD, no OSD elements displayed.
When disabling fake HD, OSD elements are shown but no HD layout (big font size)

howels, 2022-12-29:
> On BF 4.4, when enabling fake HD, no OSD elements displayed. When disabling fake HD, OSD elements are shown but no HD layout (big font size)
> 
> EDIT: my v2 googles are on v0.9.0 msp-osd

Disable fakeHD and use realHD for 4.4

Mlanaga, 2023-01-03:
On BF 4.4, fakehd=off and OSD HD=ON in Betaflight, 
You must plug your quad before your goggles to have real HD canva working isn't it ? 
If i plug the goggle first, i haven't any HD OSD working... Am i doing something wrong ...

benlumley, 2023-01-03:
Relevant copy/paste from discord, from @bri3d :

try  ```package-config set msp-osd compress_osd true && package-config apply msp-osd``` in the WTFOS CLI on the AU 
or, reboot your goggles, then reboot your AU

it's an ordering issue with FC/AU boot and goggles boot with the old way of sending OSD data. compress_osd switches to the new way and fixes it. it will be the default soon