fpv-wtf/msp-osd #88: OSD updates slowly when OSD1_ASPEED_EN output is enabled

hyperion11, 2022-12-05:
Hello!
msp-osd 0.8.0
Airunit + DJI FPV Googles V1
Ardupilot 4.4 dev and 4.3.1
When the airspeed output (OSD1_ASPEED_EN) is enabled, the OSD starts to update slowly. It is worth disabling the output of this parameter - the OSD becomes responsive.
compression enabled.
AP config:
OSD_TYPE = 5
SERIALx_PROTOCOL = 42
OSD1_TXT_RES = 1

bri3d, 2022-12-05:
Thanks for the report. I can't see at this point how this is really on the msp-osd side, just something on the Ardu side maybe.

If @yaapu has free time, maybe he wants to look into it?

DarkSoulRL, 2022-12-05:
I have the same problem. And sometimes when you turn on the OSD does not appear at all if OSD1_ASPEED_EN = 1

bri3d, 2022-12-05:
Is this isolated specifically to OSD1_ASPEED_EN, and not related to additional for example, altitude? Just trying to narrow it down to things that take the AHRS semaphore or not. 

The code that draws ASPEED looks extremely simple, it takes the AHRS semaphore, asks for the airspeed_estimate, and then draws a string generated using one of the common float scaling methods (u_scale) which is used in a million other places in the OSD.

I looked briefly at the airspeed estimator code and while it does perform some calculations, it doesn't seem that heavyweight and it is called in a lot of different places in main loops so if it were broken, I would expect much worse flight effects than just the OSD.

I think some kind of dynamic profiling might be necessary to figure out what is going on here. I really can't see how it could be on the msp-osd side, so it's definitely worth asking the Ardu folks about.

yaapu, 2022-12-05:
Hi @hyperion11 
I tested with current (latest) master, using a CubeOrange, msp-osd 0.8, DJI FPV Goggles V1 (no compression) and is really smooth.
Which board are you using?
Would you share your OSD layout (all OSDx_ params) for me to try to replicate.
In my testing I enabled all OSD panels

DarkSoulRL, 2022-12-06:
Matek H743 slim, Matek F405-CTR, arduplane 4.4 dev
When I load these options, osd stops working. After reboot FC, OSD is not displayed.
```
OSD_TYPE,5
SERIAL3_PROTOCOL,42
OSD1_TXT_RES,1
```

```
OSD1_ACC_LAT_EN,0
OSD1_ACC_LAT_X,0
OSD1_ACC_LAT_Y,0
OSD1_ACC_LONG_EN,0
OSD1_ACC_LONG_X,0
OSD1_ACC_LONG_Y,0
OSD1_ACC_VERT_EN,0
OSD1_ACC_VERT_X,0
OSD1_ACC_VERT_Y,0
OSD1_ALTITUDE_EN,1
OSD1_ALTITUDE_X,43
OSD1_ALTITUDE_Y,9
OSD1_AOA_EN,0
OSD1_AOA_X,0
OSD1_AOA_Y,0
OSD1_ARMING_EN,1
OSD1_ARMING_X,0
OSD1_ARMING_Y,13
OSD1_ASPD_DEM_EN,0
OSD1_ASPD_DEM_X,0
OSD1_ASPD_DEM_Y,0
OSD1_ASPD1_EN,0
OSD1_ASPD1_X,2
OSD1_ASPD1_Y,13
OSD1_ASPD2_EN,0
OSD1_ASPD2_X,0
OSD1_ASPD2_Y,0
OSD1_ASPEED_EN,1
OSD1_ASPEED_X,2
OSD1_ASPEED_Y,9
OSD1_ATEMP_EN,0
OSD1_ATEMP_X,4
OSD1_ATEMP_Y,0
OSD1_AUTO_FLP_EN,0
OSD1_AUTO_FLP_X,0
OSD1_AUTO_FLP_Y,0
OSD1_AVG_EFFA_EN,0
OSD1_AVG_EFFA_X,22
OSD1_AVG_EFFA_Y,10
OSD1_AVG_EFFG_EN,0
OSD1_AVG_EFFG_X,22
OSD1_AVG_EFFG_Y,10
OSD1_AVGCELLV_EN,1
OSD1_AVGCELLV_X,45
OSD1_AVGCELLV_Y,0
OSD1_BAT_PCT_EN,0
OSD1_BAT_PCT_X,0
OSD1_BAT_PCT_Y,0
OSD1_BAT_VOLT_EN,1
OSD1_BAT_VOLT_X,45
OSD1_BAT_VOLT_Y,1
OSD1_BAT2_VLT_EN,0
OSD1_BAT2_VLT_X,0
OSD1_BAT2_VLT_Y,0
OSD1_BAT2REM_EN,0
OSD1_BAT2REM_X,0
OSD1_BAT2REM_Y,0
OSD1_BAT2USED_EN,0
OSD1_BAT2USED_X,0
OSD1_BAT2USED_Y,0
OSD1_BATREM_EN,0
OSD1_BATREM_X,13
OSD1_BATREM_Y,7
OSD1_BATTBAR_EN,0
OSD1_BATTBAR_X,11
OSD1_BATTBAR_Y,11
OSD1_BATUSED_EN,1
OSD1_BATUSED_X,44
OSD1_BATUSED_Y,5
OSD1_BTEMP_EN,0
OSD1_BTEMP_X,0
OSD1_BTEMP_Y,0
OSD1_CALLSIGN_EN,0
OSD1_CALLSIGN_X,0
OSD1_CALLSIGN_Y,0
OSD1_CELLVOLT_EN,0
OSD1_CELLVOLT_X,13
OSD1_CELLVOLT_Y,10
OSD1_CHAN_MAX,2100
OSD1_CHAN_MIN,900
OSD1_CLIMBEFF_EN,1
OSD1_CLIMBEFF_X,44
OSD1_CLIMBEFF_Y,7
OSD1_CLK_EN,1
OSD1_CLK_X,0
OSD1_CLK_Y,0
OSD1_COMPASS_EN,1
OSD1_COMPASS_X,25
OSD1_COMPASS_Y,0
OSD1_CRS_HADJ_EN,0
OSD1_CRS_HADJ_X,0
OSD1_CRS_HADJ_Y,0
OSD1_CRS_HEAD_EN,0
OSD1_CRS_HEAD_X,0
OSD1_CRS_HEAD_Y,0
OSD1_CRSFANT_EN,0
OSD1_CRSFANT_X,0
OSD1_CRSFANT_Y,0
OSD1_CRSFPWR_EN,0
OSD1_CRSFPWR_X,0
OSD1_CRSFPWR_Y,0
OSD1_CRSFRSSI_EN,0
OSD1_CRSFRSSI_X,0
OSD1_CRSFRSSI_Y,0
OSD1_CRSFSNR_EN,0
OSD1_CRSFSNR_X,0
OSD1_CRSFSNR_Y,0
OSD1_CRSSHAIR_EN,0
OSD1_CRSSHAIR_X,0
OSD1_CRSSHAIR_Y,0
OSD1_CURRENT_EN,1
OSD1_CURRENT_X,45
OSD1_CURRENT_Y,2
OSD1_CURRENT2_EN,0
OSD1_CURRENT2_X,0
OSD1_CURRENT2_Y,0
OSD1_EFFA_EN,0
OSD1_EFFA_X,22
OSD1_EFFA_Y,10
OSD1_EFFG_EN,1
OSD1_EFFG_X,45
OSD1_EFFG_Y,6
OSD1_ENABLE,1
OSD1_ESCAAMPS_EN,0
OSD1_ESCAAMPS_X,24
OSD1_ESCAAMPS_Y,14
OSD1_ESCARPM_EN,0
OSD1_ESCARPM_X,22
OSD1_ESCARPM_Y,12
OSD1_ESCHAMPS_EN,0
OSD1_ESCHAMPS_X,14
OSD1_ESCHAMPS_Y,4
OSD1_ESCHRPM_EN,1
OSD1_ESCHRPM_X,8
OSD1_ESCHRPM_Y,1
OSD1_ESCTAMPS_EN,0
OSD1_ESCTAMPS_X,24
OSD1_ESCTAMPS_Y,14
OSD1_ESCTEMP_EN,1
OSD1_ESCTEMP_X,9
OSD1_ESCTEMP_Y,0
OSD1_FENCE_EN,0
OSD1_FENCE_X,14
OSD1_FENCE_Y,9
OSD1_FLTIME_EN,1
OSD1_FLTIME_X,43
OSD1_FLTIME_Y,11
OSD1_FLTMODE_EN,1
OSD1_FLTMODE_X,2
OSD1_FLTMODE_Y,8
OSD1_FONT,0
OSD1_GNDTRVL_EN,1
OSD1_GNDTRVL_X,43
OSD1_GNDTRVL_Y,10
OSD1_GPSLAT_EN,1
OSD1_GPSLAT_X,15
OSD1_GPSLAT_Y,17
OSD1_GPSLONG_EN,1
OSD1_GPSLONG_X,28
OSD1_GPSLONG_Y,17
OSD1_GSPEED_EN,1
OSD1_GSPEED_X,2
OSD1_GSPEED_Y,11
OSD1_HDOP_EN,1
OSD1_HDOP_X,0
OSD1_HDOP_Y,4
OSD1_HEADING_EN,0
OSD1_HEADING_X,13
OSD1_HEADING_Y,2
OSD1_HOME_EN,1
OSD1_HOME_X,24
OSD1_HOME_Y,1
OSD1_HOMEDIR_EN,0
OSD1_HOMEDIR_X,22
OSD1_HOMEDIR_Y,1
OSD1_HOMEDIST_EN,0
OSD1_HOMEDIST_X,11
OSD1_HOMEDIST_Y,2
OSD1_HORIZON_EN,1
OSD1_HORIZON_X,25
OSD1_HORIZON_Y,9
OSD1_LINK_Q_EN,1
OSD1_LINK_Q_X,0
OSD1_LINK_Q_Y,2
OSD1_LOIT_RAD_EN,0
OSD1_LOIT_RAD_X,0
OSD1_LOIT_RAD_Y,0
OSD1_MESSAGE_EN,1
OSD1_MESSAGE_X,12
OSD1_MESSAGE_Y,14
OSD1_NRG_CONS_EN,0
OSD1_NRG_CONS_X,0
OSD1_NRG_CONS_Y,0
OSD1_NRG_REM_EN,0
OSD1_NRG_REM_X,0
OSD1_NRG_REM_Y,0
OSD1_PEAK_PR_EN,0
OSD1_PEAK_PR_X,0
OSD1_PEAK_PR_Y,0
OSD1_PEAK_RR_EN,0
OSD1_PEAK_RR_X,0
OSD1_PEAK_RR_Y,0
OSD1_PGNDTRVL_EN,0
OSD1_PGNDTRVL_X,22
OSD1_PGNDTRVL_Y,11
OSD1_PITCH_EN,1
OSD1_PITCH_X,1
OSD1_PITCH_Y,12
OSD1_PLUSCODE_EN,0
OSD1_PLUSCODE_X,0
OSD1_PLUSCODE_Y,0
OSD1_POWER_EN,1
OSD1_POWER_X,45
OSD1_POWER_Y,3
OSD1_R_AVG_CV_EN,0
OSD1_R_AVG_CV_X,24
OSD1_R_AVG_CV_Y,3
OSD1_RC_FS_EN,0
OSD1_RC_FS_X,0
OSD1_RC_FS_Y,0
OSD1_RC_THR_EN,0
OSD1_RC_THR_X,0
OSD1_RC_THR_Y,0
OSD1_RESTVOLT_EN,0
OSD1_RESTVOLT_X,24
OSD1_RESTVOLT_Y,2
OSD1_RNGF_EN,0
OSD1_RNGF_X,12
OSD1_RNGF_Y,3
OSD1_ROLL_EN,1
OSD1_ROLL_X,45
OSD1_ROLL_Y,12
OSD1_RSSI_EN,1
OSD1_RSSI_X,0
OSD1_RSSI_Y,1
OSD1_SATS_EN,1
OSD1_SATS_X,0
OSD1_SATS_Y,3
OSD1_SIDEBARS_EN,1
OSD1_SIDEBARS_X,15
OSD1_SIDEBARS_Y,6
OSD1_STATS_EN,0
OSD1_STATS_X,1
OSD1_STATS_Y,2
OSD1_TEMP_EN,1
OSD1_TEMP_X,15
OSD1_TEMP_Y,0
OSD1_TER_HGT_EN,0
OSD1_TER_HGT_X,23
OSD1_TER_HGT_Y,7
OSD1_THR_OUT_EN,1
OSD1_THR_OUT_X,2
OSD1_THR_OUT_Y,7
OSD1_TUNED_PN_EN,0
OSD1_TUNED_PN_X,0
OSD1_TUNED_PN_Y,0
OSD1_TUNED_PV_EN,0
OSD1_TUNED_PV_X,0
OSD1_TUNED_PV_Y,0
OSD1_VSPEED_EN,1
OSD1_VSPEED_X,45
OSD1_VSPEED_Y,8
OSD1_VTX_PWR_EN,0
OSD1_VTX_PWR_X,0
OSD1_VTX_PWR_Y,0
OSD1_WAYPOINT_EN,0
OSD1_WAYPOINT_X,0
OSD1_WAYPOINT_Y,0
OSD1_WIND_EN,1
OSD1_WIND_X,2
OSD1_WIND_Y,10
OSD1_XTRACK_EN,0
OSD1_XTRACK_X,0
OSD1_XTRACK_Y,0

```

yaapu, 2022-12-06:
Hi @DarkSoulRL ,
looks like you're running a custom build, the following OSD items are NOT part of ardupilot

Unknown parameter OSD1_ACC_LAT_EN
Unknown parameter OSD1_ACC_LONG_EN
Unknown parameter OSD1_ACC_VERT_EN
Unknown parameter OSD1_AOA_EN
Unknown parameter OSD1_ASPD_DEM_EN
Unknown parameter OSD1_AUTO_FLP_EN
Unknown parameter OSD1_AVG_EFFA_EN
Unknown parameter OSD1_AVG_EFFG_EN
Unknown parameter OSD1_BAT_PCT_EN
Unknown parameter OSD1_BAT2REM_EN
Unknown parameter OSD1_BATREM_EN
Unknown parameter OSD1_CRS_HADJ_EN
Unknown parameter OSD1_CRS_HEAD_EN
Unknown parameter OSD1_CRSFANT_EN
Unknown parameter OSD1_CRSFPWR_EN
Unknown parameter OSD1_CRSFRSSI_EN
Unknown parameter OSD1_CRSFSNR_EN
Unknown parameter OSD1_EFFA_EN
Unknown parameter OSD1_EFFG_EN
Unknown parameter OSD1_ESCAAMPS_EN
Unknown parameter OSD1_ESCARPM_EN
Unknown parameter OSD1_ESCHAMPS_EN
Unknown parameter OSD1_ESCHRPM_EN
Unknown parameter OSD1_ESCTAMPS_EN
Unknown parameter OSD1_GNDTRVL_EN
Unknown parameter OSD1_LOIT_RAD_EN
Unknown parameter OSD1_NRG_CONS_EN
Unknown parameter OSD1_NRG_REM_EN
Unknown parameter OSD1_PEAK_PR_EN
Unknown parameter OSD1_PEAK_RR_EN
Unknown parameter OSD1_PGNDTRVL_EN
Unknown parameter OSD1_R_AVG_CV_EN
Unknown parameter OSD1_RC_FS_EN
Unknown parameter OSD1_RC_THR_EN
Unknown parameter OSD1_THR_OUT_EN
Unknown parameter OSD1_TUNED_PN_EN
Unknown parameter OSD1_TUNED_PV_EN

I loaded your layout and except for the unsupported OSD items is nice and smooth, no issues here on a CubeOrange running latest master, I'd say fps is around 10Hz

DarkSoulRL, 2022-12-06:
oh yes, it was downloaded from Arducustom v11. But to figure out what the problem is, I installed Arduplane 4.4 dev. OSD freezes and does not thump both there and there the same way.

yaapu, 2022-12-06:
I can't replicate it, I need to try with a lower end CPU. Did you try a different uart? perhaps one with DMA for both TX/RX?

DarkSoulRL, 2022-12-06:
I tried 2 different FC, as I wrote above. I changed the UART on matek H743, but this did not affect it in any way. I see only a dependency on OSD1_ASPEED_EN is enabled or disabled.

yaapu, 2022-12-06:
I'm sorry I can't help unless I can replicate it :-(

yaapu, 2022-12-06:
> @yaapu here my params:
> 
> OSD1_ALTITUDE_EN,1 OSD1_ALTITUDE_X,50 OSD1_ALTITUDE_Y,7 OSD1_ARMING_EN,1 OSD1_ARMING_X,10 OSD1_ARMING_Y,3 OSD1_ASPD1_EN,0 OSD1_ASPD1_X,0 OSD1_ASPD1_Y,5 OSD1_ASPD2_EN,0 OSD1_ASPD2_X,0 OSD1_ASPD2_Y,0 OSD1_ASPEED_EN,1 OSD1_ASPEED_X,0 OSD1_ASPEED_Y,9 OSD1_ATEMP_EN,0 OSD1_ATEMP_X,0 OSD1_ATEMP_Y,0 OSD1_AVGCELLV_EN,1 OSD1_AVGCELLV_X,46 OSD1_AVGCELLV_Y,11 OSD1_BAT_VOLT_EN,0 OSD1_BAT_VOLT_X,15 OSD1_BAT_VOLT_Y,7 OSD1_BAT2_VLT_EN,0 OSD1_BAT2_VLT_X,0 OSD1_BAT2_VLT_Y,0 OSD1_BAT2USED_EN,0 OSD1_BAT2USED_X,0 OSD1_BAT2USED_Y,0 OSD1_BATTBAR_EN,0 OSD1_BATTBAR_X,17 OSD1_BATTBAR_Y,12 OSD1_BATUSED_EN,1 OSD1_BATUSED_X,40 OSD1_BATUSED_Y,13 OSD1_BTEMP_EN,0 OSD1_BTEMP_X,0 OSD1_BTEMP_Y,0 OSD1_CALLSIGN_EN,0 OSD1_CALLSIGN_X,10 OSD1_CALLSIGN_Y,6 OSD1_CELLVOLT_EN,0 OSD1_CELLVOLT_X,38 OSD1_CELLVOLT_Y,11 OSD1_CHAN_MAX,2100 OSD1_CHAN_MIN,900 OSD1_CLIMBEFF_EN,1 OSD1_CLIMBEFF_X,45 OSD1_CLIMBEFF_Y,5 OSD1_CLK_EN,0 OSD1_CLK_X,0 OSD1_CLK_Y,0 OSD1_COMPASS_EN,1 OSD1_COMPASS_X,24 OSD1_COMPASS_Y,0 OSD1_CRSSHAIR_EN,0 OSD1_CRSSHAIR_X,0 OSD1_CRSSHAIR_Y,0 OSD1_CURRENT_EN,1 OSD1_CURRENT_X,40 OSD1_CURRENT_Y,12 OSD1_CURRENT2_EN,0 OSD1_CURRENT2_X,0 OSD1_CURRENT2_Y,0 OSD1_DIST_EN,1 OSD1_DIST_X,0 OSD1_DIST_Y,13 OSD1_EFF_EN,1 OSD1_EFF_X,45 OSD1_EFF_Y,4 OSD1_ENABLE,1 OSD1_ESCAMPS_EN,0 OSD1_ESCAMPS_X,10 OSD1_ESCAMPS_Y,11 OSD1_ESCRPM_EN,0 OSD1_ESCRPM_X,10 OSD1_ESCRPM_Y,9 OSD1_ESCTEMP_EN,0 OSD1_ESCTEMP_X,10 OSD1_ESCTEMP_Y,10 OSD1_FENCE_EN,0 OSD1_FENCE_X,14 OSD1_FENCE_Y,9 OSD1_FLTIME_EN,1 OSD1_FLTIME_X,0 OSD1_FLTIME_Y,14 OSD1_FLTMODE_EN,1 OSD1_FLTMODE_X,0 OSD1_FLTMODE_Y,7 OSD1_FONT,0 OSD1_GPSLAT_EN,1 OSD1_GPSLAT_X,40 OSD1_GPSLAT_Y,1 OSD1_GPSLONG_EN,1 OSD1_GPSLONG_X,40 OSD1_GPSLONG_Y,0 OSD1_GSPEED_EN,1 OSD1_GSPEED_X,0 OSD1_GSPEED_Y,11 OSD1_HDOP_EN,0 OSD1_HDOP_X,0 OSD1_HDOP_Y,0 OSD1_HEADING_EN,1 OSD1_HEADING_X,22 OSD1_HEADING_Y,1 OSD1_HOME_EN,1 OSD1_HOME_X,20 OSD1_HOME_Y,2 OSD1_HOMEDIR_EN,0 OSD1_HOMEDIR_X,16 OSD1_HOMEDIR_Y,3 OSD1_HOMEDIST_EN,0 OSD1_HOMEDIST_X,16 OSD1_HOMEDIST_Y,2 OSD1_HORIZON_EN,1 OSD1_HORIZON_X,24 OSD1_HORIZON_Y,8 OSD1_LINK_Q_EN,0 OSD1_LINK_Q_X,1 OSD1_LINK_Q_Y,1 OSD1_MESSAGE_EN,1 OSD1_MESSAGE_X,7 OSD1_MESSAGE_Y,13 OSD1_PITCH_EN,1 OSD1_PITCH_X,8 OSD1_PITCH_Y,17 OSD1_PLUSCODE_EN,0 OSD1_PLUSCODE_X,0 OSD1_PLUSCODE_Y,0 OSD1_POWER_EN,0 OSD1_POWER_X,0 OSD1_POWER_Y,2 OSD1_RESTVOLT_EN,1 OSD1_RESTVOLT_X,46 OSD1_RESTVOLT_Y,10 OSD1_RNGF_EN,1 OSD1_RNGF_X,0 OSD1_RNGF_Y,5 OSD1_ROLL_EN,1 OSD1_ROLL_X,13 OSD1_ROLL_Y,17 OSD1_RSSI_EN,1 OSD1_RSSI_X,0 OSD1_RSSI_Y,1 OSD1_SATS_EN,1 OSD1_SATS_X,0 OSD1_SATS_Y,0 OSD1_SIDEBARS_EN,1 OSD1_SIDEBARS_X,15 OSD1_SIDEBARS_Y,5 OSD1_STATS_EN,0 OSD1_STATS_X,0 OSD1_STATS_Y,0 OSD1_TEMP_EN,1 OSD1_TEMP_X,0 OSD1_TEMP_Y,17 OSD1_TER_HGT_EN,1 OSD1_TER_HGT_X,0 OSD1_TER_HGT_Y,4 OSD1_THROTTLE_EN,1 OSD1_THROTTLE_X,0 OSD1_THROTTLE_Y,2 OSD1_TXT_RES,1 OSD1_VSPEED_EN,1 OSD1_VSPEED_X,45 OSD1_VSPEED_Y,8 OSD1_VTX_PWR_EN,0 OSD1_VTX_PWR_X,9 OSD1_VTX_PWR_Y,5 OSD1_WAYPOINT_EN,0 OSD1_WAYPOINT_X,0 OSD1_WAYPOINT_Y,0 OSD1_WIND_EN,1 OSD1_WIND_X,0 OSD1_WIND_Y,10 OSD1_XTRACK_EN,1 OSD1_XTRACK_X,0 OSD1_XTRACK_Y,15
> 
> FC Pixracer

Ok, good news, well sort of, I had a Pixracer laying around and tested your layout, refresh is slow, disabling the sidebars fixed it.
The sidebars are rendered with a loop that draws individual characters, very inefficient code, but it was designed to run locally on the built in OSD not for displayport

hyperion11, 2022-12-07:
@yaapu i use plane, FC's Pixracer, Matek F405 wing, Matek H743Wing. At this time i test MSP-OSD on pixracer. I not test airspeed without sidebars. But i found too, disabling the sidebar really speeds up the OSD. Sidebars not nessesary for me

hyperion11, 2022-12-07:
i will try today. i also see some artefacts on AHI and sidebars. This ardupilot side issue or msp-osd?

hyperion11, 2022-12-07:
> @hyperion11 could you test if this fixes your issue, i.e allows to have full OSD with sidebars enabled? [pixracer_msposdfix_arduplane_4.4dev.zip](https://github.com/fpv-wtf/msp-osd/files/10174054/pixracer_msposdfix_arduplane_4.4dev.zip) It's a build of this PR [ArduPilot/ardupilot#22355](https://github.com/ArduPilot/ardupilot/pull/22355)
i test it. Enable sidebars and ASPEED. Everything looks good. No lags

yaapu, 2022-12-07:
> @hyperion11 could you test if this fixes your issue, i.e allows to have full OSD with sidebars enabled? [...]

this is great news.
@DarkSoulRL could you test as well?

DarkSoulRL, 2022-12-07:
Yes, but me need target matek h743 plane

yaapu, 2022-12-10:
@DarkSoulRL please test this version and report back [matekh743_msposdfix_arduplane_4.4dev.zip](https://github.com/fpv-wtf/msp-osd/files/10200417/matekh743_msposdfix_arduplane_4.4dev.zip)

DarkSoulRL, 2022-12-11:
> @DarkSoulRL please test this version and report back [matekh743_msposdfix_arduplane_4.4dev.zip](https://github.com/fpv-wtf/msp-osd/files/10200417/matekh743_msposdfix_arduplane_4.4dev.zip)

Yes, it looks like the issue has been fixed. At least I tested it on a table and the OSD never hung.

yaapu, 2022-12-11:
thanks @DarkSoulRL

hyperion11, 2022-12-18:
@yaapu this fix merged in master?

bri3d, 2023-01-21:
I am closing this issue, because it has been fixed in Ardupilot.

closed 2023-01-21 by bri3d