fpv-wtf/msp-osd #90: Make OSD metric symbols more... metric ;)

robertrypula, 2022-12-15:
For some reason this kills my eyes:

[image]

Maybe there are some places in the world that this is used on daily basis but in general there are some rules in metric system how to express units:

| full name  | correct when symbols are used | incorrect |
| ------------- | ------------- |----|
| meter  | m  | M, m., mtr, mtrs |
| second  | s | sec, S, SEC, sec., s., S. |
| hour | h | hr, hrs, HR, h., HR., HRS. |
| Ampere | A | Amp (used in spoken language but technically incorrect when used as unit symbol) |
| Volt | V |  |

Also metric system uses "/" for "per" as this **scales up very well for any unit**. We don't have to mix symbols with "p" character when expressing any combination of units. Also there is no logic behind expressing km/h as kph but using "/" for the other units like m/s, cm/s, mm/s (mps, cmps, mmps is like WAT? :D). Consistency is the key so it would be nice to stick to just one character like "/":

| full name | correct when symbols are used| incorrect usage |
|-----|---------|------------------|
| meters per second | m/s | M/s (M is mega, m is meter), m/sec (in metric system symbol for second is just "s"), M/sec |
| kilometer per second | km/h | kph, kmh, kmph |
| centimeters per second | cm/s | cmps (WAT?) |
| degrees per second | deg/s | dps?, degps? oh noo ;) |

Metric symbols are case sensitive. Maybe M/S is kid of okay because we might not have lowercase fonts but mixing it like M/s is not correct. It's not "Mega per second" :P

More info here https://usma.org/correct-usage-of-the-si

This lead me to the question. Is there any way where I can open those bin files in for example Photoshop and edit those graphics? This really kills my eyes and destroys experience :D

hickscorp, 2023-01-21:
A guess here - but as it is possible to use custom fonts for the OSD by placing them at the root of your SD card, and since these units are actually each just a single glyph in the font file, it is very likely that you should be able to patch your favourite font file and drop it on the SD card.

I might give it a shot this weekend, as this is also something that disturbs me - it doesn't hurt my eyes, but having to rely on my analytic brain while flying kind of breaks my flow :)

Knifa, 2023-01-21:
lmao ok, if you feel so strongly about it fair play 😄 [here's an update for material-osd](https://github.com/Knifa/material-osd/releases/tag/v0.0.4).

the primary reason was because `km/h` is longer than `kph` so it wouldn't fit in the glyph box unless it was scaled down, which means all the other units need scaled down to match. they end up pretty small on HD.
[image]
[image]

the fonts aren't owned by this repo fwiw, but a fun post none-the-less.

robertrypula, 2023-01-23:
> A guess here - but as it is possible to use custom fonts for the OSD by placing them at the root of your SD card, and since these units are actually each just a single glyph in the font file, it is [...]

Yeah I wanted to update them in Photoshop but I failed on finding the way on editing the ***.bin** file :) If googles could read PNG with transparency that would be awesome :D

vicewize, 2023-02-04:
just tell me exactly how u want it and ill update my font, im an engineet and i hate the imperial system (from usa) so i get it. ill read this threadn more throughly when i have time. :-)

robertrypula, 2023-02-04:
I'm now in the field testing all the fonts (finally it's not raining). I'll make comparison and l make a list :)

robertrypula, 2023-02-10:
Hello @vicewize 
I finally had some time to continue on this topic. Please find below my suggestions. I might be wrong in my thinking so I'm open for discussion :)

| current behavior  | expected behavior | comment |
:-------------------------:|:-------------------------:|:-------------------------:
[image] | [image] | SI uses "/" for "per"
[image] | [image] | M is mega, SI symbol for second is just "s" not sec
[image] | [image] | Inconsistency - if we use lower case charracters in other places when km should be with lowercase characters
[image] | [image] | "ft" for case consistensy and "s" for second. Since this is imperial unit I guess fps is also correct (according to Wikipedia)
[image] | [image] | Units should be upper case
[image] | [image] | Dot looks like minus, IMHO it looks weird on units especially here: [image] It's like - 0 _ 3 m
[image] | [image] | SI symbol for Ampere is uppercase "A". General Rule in SI is that we use upper case for units when name comes from a real person (like "W" for Mr. Watt, "V" for Mr. Volt, "A" for Mr. Ampere)
[image] | [image] | Here I'm not sure, upper case it's easier to read but we have inconsistency with places where lower/upper case are used
[image] | [image] | Same as above
[image] | [image] | IMHO the / is not needed as it looks now as "ON per minute" and "FLY per min" which doesn't make sense :D 

In general I found this website very helpful https://usma.org/correct-usage-of-the-si. Also Wikipedia is cool to double check the units.

I was also testing all the other fonts. It would be super cool to add CHROMA version of the VICEWIZE Italic font. You could use existing tools to overlay it over GoPro footage. Check my video at the end:

https://youtu.be/C83AlfBp-9k?t=186

[[image])](https://youtu.be/C83AlfBp-9k?t=186)