fpv-wtf/msp-osd #97: OSD file recording not working on Goggles v1

ObiVanDunobi, 2023-01-04:
Hi guys,
first of all, I want to thank you for what you do for the community.

WTFOS OSD works fine for me.
Today I was trying to set recording OSD file to sd card in Goggles V1 but I can't solve it.
I have WTFOS updated to the latest version, the required commands are entered via the CLI, but the OSD file still does not record to the SD card.
Please don't you know what the problem might be?
I also noticed with the CLI tab that I can't copy and paste the commands but have to enter them manually.
Could it be related?

Prints from WTFOS Goggles CLI are attached.

Thanks for help
[image]
[image]

bri3d, 2023-01-22:
Have you tried rebooting your goggles without USB/PC connected, to ensure the SD card is visible to the goggles OS instead of the PC?

ObiVanDunobi, 2023-01-22:
Yes, I tried several ways, but the osd file did not appear anywhere after
recording the DVR

W3srick, 2023-02-06:
OSD file recording not working on Goggles v1
I am having this same issue.

ysoldak, 2024-01-06:
Same here. Goggles V1, no .osd files.