fpv-wtf/msp-osd #101: osd artifacts inav 6

A0RU, 2023-01-11:
inav 6. Сompressed transmission is enabled.

https://user-images.githubusercontent.com/122446489/211789828-b1ffab9f-2b26-4c3f-aee3-671182381020.mp4

[image]

bri3d, 2023-01-21:
This happens when INAV is unable to schedule the serial task often enough to drain the serial buffer before the OSD task is scheduled.

Check that:
* "Custom OSD" is turned off in the goggles menu. This causes extra MSP traffic which will result in this issue.
* You are not using Soft Serial.

A0RU, 2023-01-24:
Thanks for your help. I use Soft Serial and "Custom OSD" is turned off in the goggles menu but the problem is still there.

> This happens when INAV is unable to schedule the serial task often enough to drain the serial buffer before the OSD task is scheduled.
> 
> Check that:
> 
> * "Custom OSD" is turned off in the goggles menu. This causes extra MSP traffic which will result in this issue.
> * You are not using Soft Serial.