fpv-wtf/msp-osd #102: OSD freezing

klavifon, 2023-01-14:
Hello,

I have an issue where OSD freezes when video signal becomes weak. Sometimes the OSD comes back (by that I mean it starts updating its values) and sometimes it just stays frozen. I have to disarm and unplug the quad and then it comes back. I don't know the exact number of mbps but it happens when video link is pretty weak. I'm kind of desperate because I really need an OSD elements on my long range quad and I very much rely on elements like home arrow and distance from home etc...
What could be the problem?
Thanks,
Ales

bri3d, 2023-01-21:
What flight controller software are you using?

klavifon, 2023-01-21:
At that time I was using Betaflight 4.3.2 (latest official release).

bri3d, 2023-01-22:
Interesting, thanks for the report. Were you using `compress_osd`? If not, you should try it out.

klavifon, 2023-01-22:
Thanks for the reply. Can I ask you what you mean by compress_osd? Some cli
based option?
Thanks

bri3d, 2023-01-22:
Documentation here: https://github.com/fpv-wtf/msp-osd/releases/tag/v0.7.0

And in the main README

klavifon, 2023-01-22:
So if I understand correctly I have to paste “package-config set msp-osd
compress_osd true
package-config apply msp-osd” in CLI on to the DJI Vista vtx using WTFOS
configurator?
Sorry for the stupid questions as I am just an ordinary man who likes to
fly 😊

BogieGirl, 2023-03-11:
Hi 
I have been complaining about this issue for a long time. I spoke with some of the developers....
at first I thought it was an INAV issue but then they came with INAV 6.0 RC3 that handles the OSD refresh as Ardopilot does, I also enabled the compress_osd feature and I work with WTFOS latest MSP versions. 
Yesterday I was 2km away flying with my wingman ( he has Ardupilot) and both of us encountered a video jamming situation, both of us lost video for 5 seconds. he recovered fine and I had my OSD freeze again so I had to land.
after landing, I reset the airunit ONLY and the freezing OSD recovered.

BogieGirl, 2023-07-31:
Any progress with this issue?

bri3d, 2023-07-31:
There are multiple possible ways in which the OSD can freeze:

* The DJI IP link fails to deliver UDP packets from the air side to the ground side
* The flight controller stops delivering MSP DisplayPort updates
* Some bug or issue in the msp-osd code causes it to hang or crash

If the link "unfreezes", that's a sign of the first issue, the DJI IP link stopped delivering packets. We can't do a lot about this as we can't really reach into the link parameters themselves. There might be a few things to do to try to "unstick" it, which could still be experimented with. Lowering the refresh rate could help a little.

If the link stays frozen and rebooting the Air Unit / Vista does _not_ fix it, that was an FC issue. Some Flight Controllers have serious issues with task scheduling and buffer management their UART tasks will crash when exposed to a large amount of MSP data while trying to send DisplayPort. This needs to be debugged and fixed at the Flight Controller level.

If rebooting the Air Unit and not the FC fixes the issue, then this could be an msp-osd bug. Blackbox logs from this situation may help me some, but these issues could prove difficult to debug and a better logging strategy is likely needed to figure them out end to end.

jaidiswangney, 2025-02-11:
hello good afternoon everyone,
I have the OSD freezing on the DJI Goggles v2 using the Air Unit Full, I have video but the OSD freezes.
sorry for my English, I'm using a translator