fpv-wtf/msp-osd #103: OSD not adjust to the screen after playback

CodingPoeta, 2023-01-20:
Hi guys, Thanks for what you have done for the fpv community.

I have found a little bug with msp-osd,

# OSD doesn't adjust to the screen

[image]

If I go view the playbacks when the air unit still connected, the OSD must be broken like above. The only way to recover is restart both goggles and air unit.

It's actually not a big problem because I guess nobody gonna open a playback while flying hahaha...

hickscorp, 2023-01-21:
Could this be related to https://github.com/fpv-wtf/msp-osd/issues/104 ?

I too have to restart both devices in a very specific order once it glitches.

bri3d, 2023-03-05:
This should have been fixed in https://github.com/fpv-wtf/msp-osd/releases/tag/v0.10.0 . Please re-open if it's not fixed in 0.10.0

closed 2023-03-05 by bri3d