fpv-wtf/msp-osd #104: OSD resolution glitches.

hickscorp, 2023-01-21:
I've been kindly pointed to this issue tracker.

The summary for this issue is here: https://github.com/fpv-wtf/wtfos/issues/57#event-8322127574

It seems that this is the more appropriate place for it.

Thanks @bri3d !

bri3d, 2023-01-21:
Use the `compress_osd` `package-config` setting on the air (Vista/AU) side.

t0ffi, 2023-01-22:
Same issue occur for me. OSD looks like on screenshot below: 

[image]

Preview in gogles looks like: 

[image]

I did `compress_osd` `package-config` and didn't resolve this issue. Tried to reinstall wtf.os both on gogles and air side.

hickscorp, 2023-01-22:
@bri3d I've done a couple of flights today and it seems to have worked. Thanks for the pointers! I'll leave the conversation open as it might be a relevant report @t0ffi brought into it.

t0ffi, 2023-01-22:
Problem solved. I've connected battery to drone, connected to BF configurator and adjusted OSD in "real time" and OSD elements moved to proper postion.

bri3d, 2023-01-22:
Yes, this makes sense as well. Unfortunately the overall grid size is not persistent in Betaflight, it is set by a negotiation each boot. So both the FC and VTx need to be turned on for the grid size to match. 

I'm closing this ticket as both issues are now resolved.

closed 2023-01-22 by bri3d

FurioMan, 2023-06-30:
Hi Guys,

Got a problem with one of my quads with Vista (on og air units it works well).
I see HD option in OSD menu, but I can't mark it. When I click it, it goes back to auto instantly.
In the google's I have the HD fonts and the readings from LQ and others and they are are working well, but whole HD osd is on SD grid, so showing itself in the googles, in the top left part of a screen.  I've already re-rooted the Vista, reinstalled WTFOS. Let me know what you think.

[image]

[image]

j005u, 2023-06-30:
> Hi Guys,
>
> Got a problem with one of my quads with Vista (on og air units it works well). I see HD option in OSD menu, but I can't mark it. When I click it, it goes back to auto instantly. In the [...]

Check that your UART is working both ways. This can happen if only one of the two wires is working.

One way to check in the Goggles is to see if the goggles can display your quads voltage. If it cannot then the UART connection is faulty.