fpv-wtf/msp-osd #108: Incorrect grid size in .osd file

twiescha, 2023-03-07:
__Setup:__
msp-osd: v0.10.0
Goggles V2 with auto recording on (rec_pb_enabled = true, rec_enabled = true)
Caddx Vista 0606 (compress_osd = true)

__Issue:__

There seems to be a timing problem when plugging in a new battery sometimes (while the goggles are on and have auto recording enabled, at least that is my setup) where the grid size in the osd file are not correct (30x15 instead of 60x22), even with compressed_osd.

It happened on my third pack, the first, second and fourth were ok, but I had to manually hexedit the osd header of the third flight to correct this. During the flight grid size was just fine in the goggles.

Do you think this could be fixed by overwriting the bytes for char_width/height on recording stop?

stefan73, 2023-03-18:
Confirmed on a Goggle V1 with HD OSD.
Same effect: osd file containts 30x15 and correspondingly incorrect SD font sizes. Patching bytes 10 to 14 in the osd file fixes the problem.
I think the key problem is that the OSD file gets created with the call from start_display() in osd_dji_overlay_udp.c. Assumption is that this happens before there is a osd connection established and thus the default values are used (sd osd).

bri3d, 2023-03-18:
Yes, I think this assumption is correct.

We'd need either some kind of "mode change" event system or additional frame metadata bytes in the OSD files for this, I think, where the frame mode changes after recording has begun.

Technically current clients could infer the frame size from the actual size in bytes of each frame (frame_size) but this is a huge hack. 

I'll chat with Knifa about this and see what he thinks.

GootyS, 2023-06-11:
Im having a small issue with OSD in BF where the right-hand side is not all the way to the edge of the OSD screen in the goggles. In the BF gui its correct though, I fly both Ultra and BF and Ultra OSD is working fine. Any ideas at all on this?