fpv-wtf/msp-osd #110: msp OSD is not working on Nebula Pro Air Unit

derpolygraf, 2023-04-01:
I have the following problem, my Nebula Pro Air Unit shows no msp OSD. My Setup is a Speedybee F7 V3 with BF 4.4. I have done all the things on https://github.com/fpv-wtf/msp-osd#flight-controller-setup. The standard OSD works perfectly (see photo). That is why I think that the diodes for TX and RX on the unit cannot be broken ( As in the case: https://www.youtube.com/watch?v=0UjG1ki1hJM). 

The msp OSD works on my old DJI Air Unit with my DJI goggles V1 on a Kakute F7 HDV.

What could be the problem here? Please help!
[Ticket-Berlin-Frankfurt-Oder-3067258252.pdf](https://github.com/fpv-wtf/msp-osd/files/11129196/Ticket-Berlin-Frankfurt-Oder-3067258252.pdf)

j005u, 2023-04-09:
Did you remember to also root and install msp-osd on the second vista?

Since you can see the DJI custom OSD then the UART must be physically fine indeed.

The other reason I can think this might happen is maybe you selected the wrong UART on the BF side for the Display Port peripheral. DJI's custom OSD would still work as it works by fetching telemetry from the FC using just the MSP standard.

derpolygraf, 2023-04-10:
> 
I found the problem, I did not update  WTFOS in the dji google. Now it works

closed 2023-04-10 by j005u