fpv-wtf/msp-osd #113: disable_betaflight_hd is undefined in schemaV2.json

j005u, 2023-04-09:
[image]

Looks like disable_betaflight_hd  was only added to schema.json.

Technically schema.json can be dropped, as long as package-config isn't actually used.

bri3d, 2023-04-09:
https://github.com/fpv-wtf/msp-osd/blob/main/config/airunit/schemaV2.json#L44 erm?

j005u, 2023-04-09:
> https://github.com/fpv-wtf/msp-osd/blob/main/config/airunit/schemaV2.json#L44 erm?

Erm indeed. I saw two people post screenshots with the same issue (no name or description) and presumed it's undefined.

Will dig out a vista and try to repro.

bri3d, 2023-04-09:
I can take a look too, my first thought was some kind of issue parsing the descriptor, but it looks right on a cursory glance. Weird.

j005u, 2023-04-09:
[image]

But le why..?

j005u, 2023-04-09:
Pinging @stylesuxx about the above issue.

Any idea what we're missing? Why doesn't the disable_betaflight_hd setting get rendered correctly?

I was able to repro on my Vista, but nothing useful in console and the schema file looks correct indeed.

j005u, 2023-04-09:
Derp herp, hurr durr. Totally missed it the first 3 times.

"name" - "title".

Also for osd_update_rate_hz.

benlumley, 2023-04-22:
Has been fixed (by bri, not me!)

closed 2023-04-22 by benlumley