fpv-wtf/msp-osd #118: Swap to Goggles GUI without uninstalling MSP OSD?

Revvilo, 2023-04-21:
Forgive me if I'm blind, but I've searched and can't find any mention of this.
It seems that while the package is installed, the OSD is always active and totally hides the DJI Goggle GUI, even though it can still be interacted with.
I find myself wanting to change transmission settings, camera configurations, etc while in the field. Is it possible to access the DJI Goggles interface without having to uninstall the MSP OSD package?

j005u, 2023-04-21:
That's not normal.

Could you please 1) check what fw version you are on and report back and b) restore the goggles using butter and re-install https://github.com/fpv-wtf/butter#usage

The problem should hopefully go away.

Revvilo, 2023-04-21:
Sorry, I might have caused some confusion.

Yes, uninstalling msp-osd allows me to access the DJI interface.

What I'm asking is if it is possible to switch to the goggle menus without removing the msp-osd package.

Revvilo, 2023-04-21:
Oh wait, I misunderstood. Using butter to update the goggles to firmware V01.00.0606, and then installing WTFOS again as normal along with the packages now shows both the msp-osd and the DJI interface!

Here I was thinking it was a trade-off, lmao. Hopefully this will serve useful to other travellers.

Thank you :)

closed 2023-04-21 by Revvilo