fpv-wtf/msp-osd #119: HD mode is only displaying upper-left part of the designed HD area.

AndKe, 2023-04-22:
My HD OSD format on Goggles V2 is not "HD". 
The goggles display the upper left part of what is in the HD OSD design in Betaflight Configurator (about 1/4 of the HD design-area is visible).
My WTFOS is updated (current release), I followed those instructions https://www.youtube.com/watch?v=8S3iBbVM4uo   

there are NO font files on the Goggles V2's SD card.  

 msp-osd packet installed on both Vista and GogglesV2
This is my msp-osd config on the Vista:
[image]

This is my msp-osd config on the goggles:
[image]

This is my OSD (work in progress) 
[image]

and this is about the part of the OSD I actually see in the goggles:
[image]

BF 4.4.1
I have used this Preset:
[image]

and verified 
set osd_displayport_device = MSP
set vcd_video_system = HD

additional information: I see OSD, I see the data change.

kf4mot, 2023-04-22:
After setting up several quads with no issues I've got one that's stuck in PAL mode like what your seeing here. There's been at least several of us on Discord with this same issue with no common cause pinpointed yet from what I've seen.

AndKe, 2023-04-22:
@kf4mot one .. vista or one Goggles, or maybe flight conroller? 
can you confirm which device the problem is on? 
is there a  specific Discord thread about this issue?

AndKe, 2023-04-22:
despite this sad result, I upgraded another one to BF4.1.1  (both were 4.3.1 before)
both have Vista's with the same firmware/config.

What is VERY interesting - is that one of them (MiniApex) works perfectly fine. with HD OSD, as expected.
The other one (GEPRC)- not.
I posted the first(non-working) config above,  here is the one that works fine:
[BTFL_MicroApex_4.4.1.txt](https://github.com/fpv-wtf/msp-osd/files/11301543/BTFL_MicroApex_4.4.1.txt)

All is conformal coated, soldered, then coated solderpoints - otherwise I would consider swapping the vistas to check if the issue is in the Vista - or the flight-controller.

kf4mot, 2023-04-22:
I have one vista stuck in PAL mode. Several other vistas working great in HD mode. All with the same V2 goggles.
My particular problem FC is a TuneRC F405, but I have other F405's working.
There's not one thread on discord, but several. None of them lead to solutions that I've seen though before they fizzle out after getting the same common recommendations for what to check. Or maybe people just stop posting once they solve the problem?
I haven't been following close lately though. I'm a bit over it and moved on at the moment.

AndKe, 2023-04-22:
Additional information:
I added osd_canvas* settings that were in the working quad. That did not help.
I also uninstalled wtfos, and re-rooted/reinstalled all on Vista - with no change in result.  
Also tried to switch to another, virgin OSD profile.

AndKe, 2023-04-24:
I just used the autodetect for firmware, (which includes HD)
[image]

AndKe, 2023-04-24:
@benlumley it's kind of done - on at least two occasions: I reverted temporarily to 4.3.1. to   compare it's behavior (PRID/rate related) - then back to 4.4.1 , and then ran a core installation to report missing blackbox options on this controller (and installede 4.3.1 to verify).  - and I am again on 4.4.1

AndKe, 2023-05-06:
So I finally figured it out:
MSP-OSD works with only RX of the Vista receiving MSP data.
If only TX pin from Vista to the flight controller is not connected - then the symptoms are:
-old/lowres OSD shows up as expected.
-No HD-size OSD
-Drone battery voltage in the native DJI OSD is displayed as "N/A"
-The Vista stays in low-power mode after arming.

The reason:  I bought a use GEPRO Thinking P16 that was incorrectly wired, the correct RX pad was never soldered on, and the TX from Vista were connected to (who-knows-what)     I could not find the pinout/schematic for this particular GEPRO 411 AIO board.
[image]

Please add those symptoms to troubleshooting FAQ - it may save a lot of time for somebody else.

kf4mot, 2023-05-09:
Great find! That was my issue too!
I assumed that since the OSD was partially working at least that the uart was fine.
Turns out that only half the uart was operational. My uart rx was set for sbus on the FC, so only tx data was getting through.

j005u, 2023-05-09:
Ideally we might want to detect when the command to identify the FC fails (meaning _something_ is up with the UART, unless the user is using some really weird FC) and display a toast on screen.

Have seen quite a few users with the same issue just lately.

I've added this to https://github.com/fpv-wtf/wtfos/wiki/Troubleshooting, it should probably be duplicated to the msp-osd README as well.

materia79, 2023-11-06:
> So I finally figured it out: MSP-OSD works with only RX of the Vista receiving MSP data. If only TX pin from Vista to the flight controller is not connected - then the symptoms are: -old/lowres OSD [...]

I tried to run msp-osd having only the RX from a Vista connected with the TX3 on a Mamba Basic F405 MK3.5 but could not get the OSD showing up at all. Is there any way to have the msp-osd being connected only to one TX of the FC and nothing else?

It seem to not work anymore and I tried any possible setting incl. to disable HD on the Vista in the wtfos configurator for msp-osd...

UART 1, 2 & 4 will be already used for CRSF RX, GPS and Serial through Crossfire since this build is going to be a waterproof build and UART 3 is only having TX3 exposed on the FC PCB...

Oh yeah and I tried all this with INAV 6.1.1

Thanks for any help or suggestion