fpv-wtf/msp-osd #120: msp-osd not showing up (INAV 6.0)

Momo-GitHub-1, 2023-04-25:
**Devices/setup:**
- FC: Diatone Mamba F722 APP
- Goggles V2, FW v1.00.0606
  Trace ID: 32a1bdf7-c8cd-4cb9-b65f-b86308c1c0aa
- DJI Airunit FW v1.00.0606
- INAV 6.0
- rooted v2 goggles/airunit, installed wtfos and packages (msp-osd)

_without msp-osd:_ 
- standard INAV "DJI FPV VTX" serial port config for custom OSD is working 
--> serial port wiring seems correct

**Describe the issue**
- INAV: switched to "MSP Displayport", enable HD OSD, "DJIWTF"
--> msp-osd seems not to work as no OSD elements are showing up 

_Please give debug hints:_
- how to debug if MSP messages are received in the air unit msp-osd part?
- how to debug if there is according data inflow in the goggles SW?

relaxibus, 2023-05-08:
Did you verified UART TX/RX wires?  May be swap them. It sounds like no UART data are received by the VTX.

Momo-GitHub-1, 2023-05-08:
UARTS are OK, as there ist the reduced OSD Info shown when I set this UART to "DJI VTX"