fpv-wtf/msp-osd #121: How to read .OSD files on Windows?

relaxibus, 2023-05-07:
Hi, I tried to open a .OSD file on Windows using Notepad++, however it shos only "NULL" fileds. Is there a way to explore the OSD file? There are few reasons:
- Get the latest GPS locations in case you lost your plane/copter
- plot GPS trace as a GPX of KML file on various tools
- to use Inav Log Analyzer to analyze your telemetry if you don´t have a onboard SD card

Thanks for any advice!

j005u, 2023-05-07:
.osd is a binary format that represents a fixed grid of glyphs (FC font characters/icons) laid out on the screen, there is no structured telemetry data.

It would be possible to extract data from each OSD frame from a certain position and translate it back to machine readable coordinates, however the precision would be exactly as shown on screen during flight and the correct x,y coordinates of the GPS position would need to be entered manually or someone would need to make a tool that scans for the Betaflight GPS icon in the frame and figures out where the coords are like that.

In other words, msp display port is really bad for this purpose due to it's very nature and we haven't put any real effort into making tools that could extract telemetry data from it because of this. Someone could hack something together, but it would be fiddly and work on a case-by-case basis.

If the plane/copter is really lost, you can always render out the osd overlay on fpv.wtf and get the last coordinates that way.

We've discussed recording GPS coordinates separately as proper telemetry for this but there are bandwidth concerns on the main FC to VTX uart when trying to do this alongside the OSD, unless we limit the update rate significantly. Another option we've discussed would be to just let the VTX be a blackbox peripheral for the FC via a separate UART. Neither is implemented right now.

relaxibus, 2023-05-08:
Thanks for clarification. However it would be good to have maybe a third file with just 1 to 0.5 Hz GPS data. There is no need to have a high GPS refresh rate.

j005u, 2023-05-08:
That would essentially be the first option of the two I mentioned at the end of my post.

Currently it's a bit complicated to do cleanly, as we need to be able to route different MSP messages between the DJI GUI, msp-osd and now additional services (like this GPS telemetry logger), but we are working towards that as time allows.

I think we can keep this issue open until then.

relaxibus, 2023-05-12:
Yeah, so we can share ideas about this. What about an online .OSD to CSV (or GPX/KML) converter? Beside the OSD Overlay Tool such a converter tool? I guess this would be helpful for many users?

j005u, 2023-05-12:
Entirely possible, but with all the caveats above in mind we haven't chosen to put any effort into this from the core team as it makes more sense to work towards proper telemetry recording.

With that said, if someone wants to take it on and has specific questions, we'll gladly assist.

The best place to start would be to look at msp-osd code to figure out the .osd binary format. There should be _some_ documentation on a GH issue here somewhere as well, but I'm not sure how up to date it is.

David-OConnor, 2023-08-20:
Hijack! How did you get the OSD file? TY!

relaxibus, 2023-08-21:
is generated automatically in the googles.

David-OConnor, 2023-08-21:
Which goggles/Vtx? Where on the SD? Ty

sbvualo, 2023-09-18:
I've wrote simple msp-osd parser https://github.com/sbvualo/pymsposd