fpv-wtf/msp-osd #123: Add configurable log levels

j005u, 2023-05-13:
In order to help debugging user issues like osd recording not working or canvas size negotiation failing we should add configurable debug levels to msp-osd.

I imagine we want at least two levels (+ disabled) because anything that fires once per frame should probably be it's own level. 

Ideally with a separate log file in /opt/var/log/msp-osd/msp-osd.log but we can also deal with printf via dmesg output worst case. We can add a dependency for logrotate (once added to the mirror repo) and a config file to make sure it never gets out of hand.

Specifically I'm trying to find out why kelso88#0828 from Discord can't get osd to record, even though the setup appears valid otherwise.

j005u, 2023-05-13:
PS: there was earlier development on the configurator to allow exporting of package logs via the UI: https://github.com/fpv-wtf/wtfos-configurator/issues/106

It was not shipped however.

benlumley, 2023-05-14:
no disk space usage issues to worry about i presume; we have enough headroom there that it should be ok ?

I think as a start....

new log function
replace all toast calls with log calls on goggles
log can write to file; on goggles can also call toast as configured

and then sprinkle logging calls around all the init / negotiation type stuff

j005u, 2023-05-14:
I have 2GB free on my V2 Goggles blackbox, so yeah, it should entirely be fine. We'll need to add logrotate or something similar for sure, otherwise eventually it'll still blow up. And definitely lets not default to any per frame logging (except maybe fatal errors I spose).

I'll take it upon myself to have logrotate available in the repo for a dep soon.

Makes sense to have all toasts logged, but I'd leave it to the maintainers to discuss how this makes sense. i.e. replace toast calls with logging calls with a flag to also show a toast, or have the toast function _also_ call the log function.

I imagine we might want to dump _more_ data to the log than necessarily show in the UI.

j005u, 2023-05-14:
Actually, I wonder if it'd make sense to use the android log functions and DJI-s existing blackbox logger to write to disk and take care of rotation etc. I'll look into it shortly, my main concern is being able to distinguish msp-osd logs from general glasses logs in a clean way.