fpv-wtf/msp-osd #128: OSD Freeze when video signal is weak

BogieGirl, 2023-07-31:
the issue is old, I have been complaining about it for a year. 
Happens when video quality ( signal) goes low , then video link goes back again to normal - the OSD freezes. 
resetting power to FC is needed to unfreez OSD, power reset to air unit doesn't unfreeze.

the following is set up:

package-config set msp-osd compress_osd true  
package-config apply msp-osd

package-config set msp-osd cache_serial true 
package-config apply msp-osd

bri3d, 2023-07-31:
> resetting power to FC is needed to unfreez OSD, power reset to air unit doesn't unfreeze.

This is your answer here, then, the issue is in the FC side.

BogieGirl, 2023-08-01:
> 

I don't think I have an answer since the problem starts when bit rate goes down or when a sudden video loss appear for a short duration. after that OSD freezes , tries to recover without luck.