fpv-wtf/msp-osd #139: Add Betafglith Goggle* and VTX* elements?

patternseek, 2023-10-27:
Hi,
In BF 4.4 (maybe earlier too, I don't know) there are OSD elements such as Goggle link quality, Goggle system warnings, VTX bitrate, VTX delay and VTX temperature that it would be nice to be able to place where we want on the screen. I realise some of these are already provided by the Goggles but for example I'd really like to have VTX bitrate and delay visible closer to the center of the screen so that it's easier to keep an eye on them when flying near the edge of reception.

Is it possible to incorporate these or is it a limitation of the data available to BF and/or wtf-os?

Thanks!

j005u, 2023-10-29:
This is certainly possible, however the original devs have been rather busy with other matters recently and are mostly in maintenance mode currently. The upcoming BF 4.5 release requires focusing on expanded font support.

We welcome any PRs to push this or other features forward and are happy to assist new developers on Discord if they need any help.

patternseek, 2023-10-29:
I think it's outside my skillset unfortunately,  I've never written any C. Is it ok to leave this open for later?

j005u, 2023-10-29:
Yes, let's leave it open for reference.