fpv-wtf/msp-osd #143: Recording PX4 MSP OSD

UnicornMoe, 2024-01-25:
The Goggles V2 only record MSP OSD (generate a .osd file) when the airunit is wired to a Betaflight flight controller, not when wired to my PX4 flight controller in my plane.
Is this a known issue? Is there a way around it?

bri3d, 2024-02-19:
I suspect that PX4 is not sending a flight controller variant response to the MSP flight controller variant query. This is required for OSD recording as it's encoded into the recorded file to allow the overlay tool to select the correct font package on the other end. 

I haven't looked at the PX4 MSP OSD implementation much - I think you're actually one of the first people who has ever posted about it, but I think I have a compatible board somewhere so I can take a look eventually.

ninja-hu, 2024-08-10:
I am trying to make the MSP OSD implementation work with my pixhawk and PX4 but I don't see the OSD parameters in QGroundControl after installing the default firmware file. How can I access those parameters?

ninja-hu, 2024-08-10:
Some updates - After compiling the PX4 firmware locally and enabling OSD : https://docs.px4.io/main/en/frames_plane/reptile_dragon_2.html#msp-osd-module I can see the MSP_OSD_CONFIG parameter in QGC and set it to TELEM2. However, I still fail to get the message on the goggles. Are there any other steps involved? (I am using fmu-v6c and version 1.14.3.

SolderSyntax, 2025-07-16:
This issue is a bit stale, but PX4 recently re-implemented their MSP OSD and it might be working again, it's worth testing and closing this issue if it works.

ninja-hu, 2025-07-17:
Tried on the AVATAR HD VTX V2 and AVATAR HD goggles, but no luck. This is after using compiling the main branch which should have the fix: https://github.com/PX4/PX4-Autopilot/pull/24695. 

For testing I just have a Pixhawk 6X without any other sensors or peripherals attached. It does seem that the messages should show up even without arming, based on the pull request. Checking the console shows the following. I am not sure if  the "No VTX configuration available" error is related or not. Appreciate any help or guidance!

[image]

maganaluis, 2025-08-25:
I have a Pixhawk 6C, I was not able to get it to work with a walksnail moonlight vtx. I tried flashing ardupilot, but I still did not get any osd to show up, so could be an issue with the Pixhawk 6c/x ?