fpv-wtf/msp-osd #144: Tx uplink Power OSD element is not working

lmc2, 2024-02-17:
the "Tpwr" or also called "Tx uplink Power" OSD element is not working in Betaflight.
Vista and V1 Googles, latest WTF firmware

bri3d, 2024-02-19:
The WTFOS msp-osd has no way to know what elements it is displaying; it just draws what the FC sends it. 

A few things to check:

* Is your Tx/Rx system configured in a way that Tx Uplink Power should work? (for example, on ELRS, it doesn't work with `Hybrid` switch mode?).
* On Betaflight 4.5, try `set displayport_msp_fonts = 0,0,0,0` to make sure that warnings don't get rendered with an unsupported font page. But if you tried 4.4.2, that's probably not it.

lmc2, 2024-02-19:
Thank you so much !!!!!! i totaly forgott about the switch mode! it works :)

closed 2024-02-19 by lmc2